dynawo/dynawo #85: Error message when there is a typography mistake in a Modelica model

MarianneSaugierRTE, 2019-02-28:
Today, when there is a typography error in a Modelica model (let's say for example that there are two ; at the end of a parameter or a variable definition instead of one), we get the following error while trying to launch a job:
`Error: Class Dynamo.AAA.BBB not found in scope CCC.`
It would be good to have a more explicit message.

gautierbureau, 2019-03-01:
I also had a problem when doint import ModelTest and creating mode ModelTest, copy/paste error from my part and creating a recurisve call in the compiler resulting in seg fault for the compiler... 

My idea would be maybe to check if OMC provides a syntaxic analysis of a .mo that we could run before installing each .mo file in the compilation toolchain.

rosiereflo, 2019-04-10:
In fact the error is hidden in the middle of the model compilation log file. I am going to improve the final error message to highlight it.

closed 2019-04-12 by rosiereflo