dynawo/dynawo #277: Fix OpenModelica checkout

gautierbureau, 2019-05-24:
They changed the way how their repo is managed, before OMCompiler was a submodule and now it is a subtree. So when we were doing git checkout it ended in OMCompiler deletion and some weird things. I decided to directly clone the right tag with a shallow clone. Some problems can happen with submodules (3rdParty) inside OMCompiler because of the shallow clone, it is because it not possible to checkout a sha1 that is not attached to a name (and OMC did not do properly tags in their repo...), the git error is "Server does not allow request for unadvertised object". The solution is to adapt to a good depth with `--depth` option.

closed 2019-05-27 by rosiereflo

marcosmc, 2019-06-13:
After this fix, we are not able to do a proper checkout. We receive the following error:

> Cloning into '/home/dynawo/dynawo/OpenModelica/Source'...
> remote: Counting objects: 252318, done.
> remote: Compressing objects: 100% (55606/55606), done.
> remote: Total 252318 (delta 194966), reused 249336 (delta 192832)
> Receiving objects: 100% (252318/252318), 188.58 MiB | 1.54 MiB/s, done.
> Resolving deltas: 100% (194966/194966), done.
> Note: checking out '38af1b628018a0468919e458e8299242703cd867'.
> 
> You are in 'detached HEAD' state. You can look around, make experimental
> changes and commit them, and you can discard any commits you make in this
> state without impacting any branches by performing another checkout.
> 
> If you want to create a new branch to retain commits you create, you may
> do so (now or later) by using -b with the checkout command again. Example:
> 
>   git checkout -b <new-branch-name>
> 
> Submodule 'OMC' (https://openmodelica.org/git-readonly/OMCompiler.git) registered for path 'OMCompiler'
> Cloning into '/home/dynawo/dynawo/OpenModelica/Source/OMCompiler'...
> remote: Counting objects: 2165, done.
> remote: Compressing objects: 100% (1637/1637), done.
> remote: Total 2165 (delta 711), reused 1194 (delta 493)
> Receiving objects: 100% (2165/2165), 8.14 MiB | 2.98 MiB/s, done.
> Resolving deltas: 100% (711/711), done.
> remote: Total 0 (delta 0), reused 0 (delta 0)
> error: no such remote ref 5ccf9cba53aeb129367431210527aa3eb8bf1c4d
> Fetched in submodule path 'OMCompiler', but it did not contain 5ccf9cba53aeb129367431210527aa3eb8bf1c4d. Direct fetching of that commit failed.
> Git clone of OMCompiler in /home/dynawo/dynawo/OpenModelica/Source failed.
> OpenModelica source code is not well checked-out. Delete your current source folder and relaunch command.

gautierbureau, 2019-06-13:
I would recommend deleting OpenModelica source folder to start from scratch because I am not enable to reproduce the error with a clean install. If it does not do the trick you can send me an email at gautier.bureau@rte-france.com to give me more precsision on your OS and other to see if I can reproduce the problem.