dynawo/dynawo #571: Improve performance on medium sized networks

rosiereflo, 2019-12-09:


adriguir, 2020-01-29:
@rosiereflo we can perhaps close this issue, at least temporarily

rosiereflo, 2020-01-29:
Not yet, I still have one PR to create on this topic

adriguir, 2020-05-06:
A few more ideas to improve performances that came to my mind in the last few days (they need to be discussed and if you find one of them interesting, we can open a specific issue). Once they are treated or given up, we can tick them with a kind of status and the link to the proper issue or discussion:

- [ ] Regarding initialization, taking advantage of the fact that we are doing a lot of treatments on the same model to avoid searching in maps or vectors several time. For example, in initModelsDescription, in the parts related to the referenceIIDM and the static references, we find the same component interface several times, whereas we know we are working on one component interface. 

- [ ] Regarding f calculations, at the beginning of each time iteration (in the SIM solver), we evaluate f with the current y and y' values to scale the norms and so on, and that also gives us the first NR result. However, as long as we do 0-order prediction (keeping the last y values from the last time-step iteration as the first y values for the current time-step iteration), the residual values from the last evalF are still valid for algebraic equations. Only differential variables need to be evaluated again (this should avoid 80% at least of the evaluations for the first iteration of each NR).

- [ ] Regarding derivatives evaluations in the network model, as long as the topology is the same, I have the impression that we don't need to recalculate them (at least on many network components). 

- [ ] Regarding f evaluations, when we need to calculate the J in a Modelica model, we need to do a special f evaluation (evalFAdept). If we are at the last evaluation before a potential J calculation/update, we can perhaps directly use this evalFAdept for the evaluation of F instead of the classical evalF, to avoid doing it twice.

- [ ] Regarding J evaluations, when we do a reinitialization, we calculate the whole J and then we delete the columns and rows we don't want. Going back to the time-domain problem, we then (in some cases) recalculates the J sometimes. Perhaps we can keep the J calculated by the reinit part and use it (especially if we factorize at each iteration for Algebraic_Mode_J_Update).

- [ ] Still my old idea on root methods to differentiate between three methods (normal evalG, evalGTime for g depending on time and eventually other things and evalGNotTime for g not depending on time). Easy to differentiate between the three of them in Cpp models, more challenging on Modelica models (nevertheless, I would say that we can find if the equation is time-dependant using the json dependency file provided by OM). Once we have this, we can do evalG normal when we are in normal cases, evalGTime when we don't evaluate y anymore (steady-state seems to be reached) and evalGNotTime when we do reinitialization (because time won't have changed -> not so sure if there are also other conditions). Could be also propagated to z.

- [ ] Avoid doing useless treatments. For example, I believe (need to be double checked) that we add things to the timeline even if we won't output it at the end and this is only the external file writing that is inhibited. Checking all the treatments would be a good idea.

- [ ] What about initially disconnected components? Should we keep them in the problem? Should we have an option to specify this in the jobs (won't be the most of the simulations, but could be some specific ones)? The same goes for not main synchronous area components. It does additional variables and equations to get through during the whole problem.

- [ ] This is more something specific for simplified simulations in the long-term but I mention it here not to forget about it. Constraints calculations should be optional and we should have the possibility to do it only for the last time step. It also means that evalG and evalZ that are there only for outputting constraints should be separated for the other ones (not really needed to do them at each g iteration but only once at the end after z stabilization I would say -> this could lead to small improvements also for voltage stability simulations, one example is the bus evalG method).

@rosiereflo @gautierbureau @marcochiaramello @cossartque @RomainLosseauRTE @MarianneSaugierRTE fyi and open to comments and discussions. Please also enrich if you have other ideas in mind.

closed 2020-05-13 by rosiereflo