dynawo/dynawo #707: Create a mechanism to update the yType and fType during the simulation

adriguir, 2020-03-09:
In case the equations for one model are completely changing during the simulation (for example from differential to algebraic), it is necessary to update the yType and fType vectors.
It would be nice to do it only on the models where it could happen, in order not to deteriorate the performances on large test cases

rosiereflo, 2020-03-23:
@adriguir question: what happens to the modes when the equation changes its type during simulation?

adriguir, 2020-03-23:
That's a good question.

I would say that normally speaking, the conditions to trigger the mode won't differ. Nevertheless, I suppose (I'm not sure anymore but if you ask this question it should be the case) that we use the fType to flag the mode change in the Python scripts and that it should then be updated if the fType is updated during the simulation. Indeed, if the equation becomes differential, if a mode change occurs later on during the simulation, it should raise a differential mode flag.

If the problem is the one guessed above ;), one possible implementation would be instead of transforming the fType directly to a mode flag into the script to use an auxiliary method doing the association between fType and mode flag at a particular instant thus using the current fType.

rosiereflo, 2020-03-23:
You guessed well! 
For CPP models this will be done manually, so this is a point of particular attention.

adriguir, 2020-03-25:
Modify the unit test for centos6 machine.

closed 2020-03-26 by rosiereflo