dynawo/dynawo #728: Name harmonization in transformers models

adriguir, 2020-03-18:
@MarianneSaugierRTE @RomainLosseauRTE @cossartque I'm currently creating an IdealTransformerVariableTap model and when modifying the package.order, I noticed that we have:
TransformerFixedRatio but TransformerVariableTap.

Does it seem fine for you? Should we harmonize to TransformerFixedTap and TransformerVariableTap? Or TransformerFixedRatio and TransformerVariableRatio?

All propositions are welcomed.

MarianneSaugierRTE, 2020-03-18:
It is not that disturbing for me given that the first one has a ratio fixed as a parameter and the second one has a ratio that is a variable and linked to the tap that is another variable. The main info is for me fixed ratio for the first one and variable tap or variable ratio for the second one

adriguir, 2020-03-23:
If no other opinions, I will consider the answer from Marianne and let it this way. Thanks!

closed 2020-03-23 by adriguir