dynawo/dynawo #739: [Modelica Models]: Do not impose the same order in the equations in the different statements of an if

rosiereflo, 2020-03-23:
Modelica :
 if condition then
    Eq1a  //algebraic
    Eq2a //differential
else
    Eq2b //differential
    Eq1b  //algebraic

results in 2 differential equations in the final model as Eq1a (resp. Eq2a) is grouped with Eq2b (resp Eq1b) and as we consider that if one branch of a if is differential all branches are.

This development as a dependency on Ticket #707.

adriguir, 2020-07-26:
@rosiereflo I think we can close this one, can't we?

rosiereflo, 2020-07-27:
The issue is still there and the fix is waiting to be commited (I waited for the release branch to be created as I didn't want this to be in the release), so I propose to not close it. I will probably be able to properly test the dev and merge it this fall.

closed 2021-03-23 by adriguir