dynawo/dynawo #794: staticRef and modelica layers

MarianneSaugierRTE, 2020-04-07:
Are modelica layers well handled for staticRefs? @adriguir @rosiereflo 

Example on the staticVarCompensator:
The following doesn't wotk:
<staticRef var="SVarC_injector_PInjPu" staticVar="p"/>

If we define a variable PPu in SVarC and write PPu = injector.PInjPu in the mo file, the following works:
<staticRef var="SVarC_PPu" staticVar="p"/>

adriguir, 2020-04-07:
Honestly, I don't know at all and I have never gone into details in this part of the code.
What I can say is that it is the good syntax and it seems to be there in the desc.xml file: SVarC_injector_PInjPu with this wording.

rosiereflo, 2020-04-09:
Let me take a look

rosiereflo, 2020-04-09:
Would you have a testcase?

MarianneSaugierRTE, 2020-04-09:
working on it ;)

MarianneSaugierRTE, 2020-04-09:
sorry... it seems that my problem was linked to something else: it works fine!

closed 2020-04-09 by MarianneSaugierRTE