dynawo/dynawo #827: Rewrite not running equation for transformers

MarianneSaugierRTE, 2020-04-28:


MarianneSaugierRTE, 2020-04-28:
We would like to write
` 
if  running 
...
else
  terminal1.i = Complex(0);
  terminal2.i = Complex(0);
end if
`
But then we got some error in test cases using preassembled models with load + transformer.

MarianneSaugierRTE, 2020-04-28:
Trying to also change the condition in the load model as follows doesn't work:
`if running ... else PPu=0; QPu=0; end if`

MarianneSaugierRTE, 2020-04-28:
The error is a fail of solver at initialization stage

rosiereflo, 2020-04-29:
Is there a branch I could use to make sure this is not caused by the model compilation?

MarianneSaugierRTE, 2020-04-29:
Yes, I just pushed it: 827_running_tfo_load. Thanks for having a look!

rosiereflo, 2020-04-30:
I don't see anything weird in the generated files. For some reason kinsol is not able to solve the system with this form (I didn't have a deep look at what happens in the solver yet). @adriguir any hint?

adriguir, 2020-04-30:
I put all my hopes and expectations onto the #739 ;(
I haven't yet had a look to it, will let you know once done

rosiereflo, 2020-04-30:
Yes I had the same idea, but in this case all equations are ALGEBRAIC :'(

adriguir, 2020-04-30:
What is strange is that anyway, this branch of the if shouldn't be evaluated. Or do we have an event disconnecting the transformer/load?

MarianneSaugierRTE, 2020-04-30:
Exactly!!! The failure occurs at init... and normally all running=true....

MarianneSaugierRTE, 2020-05-04:
Eventually it is the expected behavior to explicitly set the voltage to zero for any network injection

closed 2020-05-04 by MarianneSaugierRTE