dynawo/dynawo #842: Complex parameters not appearing as readOnly

adriguir, 2020-05-04:
When you have a complex attribute that is a final parameter, it seems we are still obliged to fill it in the .par file, which doesn't seem to be a correct behavior.

For example, if we modify the BaseGeneratorSimplified_INIT in the following way:

adriguir, 2020-05-04:
@MarianneSaugierRTE @rosiereflo what is your opinion? Shouldn't they be internal parameters?

MarianneSaugierRTE, 2020-05-05:
I don't know what it means in terms of implementation but yes it should work for s0Pu same as PGen0Pu

rosiereflo, 2020-05-06:
You are both right, they should be internal and not needed in the par file. Any branch that I can use to investigate?

adriguir, 2020-05-06:
I pushed the branch 842_TestFinalParameters to provide a test case.

closed 2020-05-07 by adriguir