dynawo/dynawo #853: Parameters curve utility

adriguir, 2020-05-06:
I was wondering if the parameters curve functionality is needed or not. From my point of view, I don't understand this need to plot parameters as they are constant throughout the simulation so curves don't seem to be the right tool to see their values.

In addition to this, @rosiereflo did the job to recover the parameters values in a nice way, which can still evolve if necessary, in this development #840 . 

I would propose to delete this functionality, except if someone already used it once in its life and found it useful @MarianneSaugierRTE @RomainLosseauRTE @rosiereflo @cossartque

rosiereflo, 2020-05-07:
I used it a few times when the parameter is a threshold to compare the difference of the value and the threshold graphically

adriguir, 2020-05-07:
Me too in fact :) now that you talk about it.
And it is quite useful indeed for this use.

closed 2020-05-07 by adriguir