dynawo/dynawo #863: Improve tap and phase shifter Modelica models control part

adriguir, 2020-05-10:
This issue targets at modifiying the tap and phase shifter control implementations in Modelica to decrease the number of root functions  present into the model.

As of today, the number of g is equal to 43 for the single TapChangerAutomaton while the number of g is equal to 6 into the equivalent Cpp Model (DYNModelRatioTapChanger.cpp).
An intermediate number preserving the higher readability of the Modelica code is possible and will be implemented.

This should help improving a bit the performances.

adriguir, 2020-05-10:
@rosiereflo I have a strange issue here that I don't manage to understand. If you could have a look to it, it would be nice!

When doing my initialization with the LoadOneTransformerTapChanger model, if tap0 for the tap-changer is 10, the initial discrete value for tap is 0. I haven't found out what was going on. A few hints:

- I pushed a branch to reproduce the problem (863 ControlTapChanger).

- When I launch the IEEE14_TapChangerAutomaton, where the tap changer automaton is using a specific preassembled model and connected through the dyd to the transformer model, I don't get any problem.

- When I launch the DisconnectGravelines6 test case from the voltage-stability repository, I get this strange behavior at initialization (see DM_ABBEVY411 and attached file)
[dumpInitValues-DM_ABBEV4Y411.txt_withDevelopment
[dumpInitValues-DM_ABBEV4Y411.txt_withoutDevelopment](https://github.com/dynawo/dynawo/files/4606082/dumpInitValues-DM_ABBEV4Y411.txt)
](https://github.com/dynawo/dynawo/files/4606080/dumpInitValues-DM_ABBEV4Y411.txt)
.

rosiereflo, 2020-05-11:
This is caused by an issue in the initialization : during the Z propagation in local initialization, the tap value takes its pre value which is not properly initialized. Let me fix it.

rosiereflo, 2020-05-11:
solved by #868

closed 2020-05-15 by MarianneSaugierRTE