dynawo/dynawo #874: Handle the running status propagation

adriguir, 2020-05-12:
In the current situation, each individual component has its own switch off and running variables that are independent from each other. In preassembled models, these variables can't be connected together without getting in trouble with the number of equations/variables.

It means that if a LoadOneTransformerTapChanger is disconnected, only the load part will really be disconnected.

From a functional point of view, it means we don't really manage the disconnection effect, even if in today's cases, it won't have a deep effect on results (except for auxiliary in machine models but they should stay connected so the behavior in this case is fine).

From a performance point of view, it means we keep three discrete variables (transformer_running, tapChanger_running and load_running) while we can have only once.

Nevertheless, I don't have any idea on how to implement it in a proper and elegant way: the only solution I have in mind would consist in deleting the extends SwitchOff part in most of the models but this will make them difficult to use in standalone mode.

@MarianneSaugierRTE @RomainLosseauRTE if you have any suggestion on the implementation, please feel free to provide it.

@rosiereflo fyi

MarianneSaugierRTE, 2020-06-12:
With today's implementation we can't link the running variables together because they only depend on the different switchOffSignals variables.

What we should do to better deal with component disconnection is to link the switchOffSignals variables together. As you already mentionned here-above, we have trouble compiling the models if we do that.

Let's have a closer look on LoadOneTransformer model.

- If a add the following line in the preassembled model I get the error: The model has 31 equation(s) and 32 variable(s).
`<dyn:connect id1="load" var1="switchOffSignal1" id2="transformer" var2="switchOffSignal1"/>`

- If a add the following line in the preassembled model I get the error: The model has 33 equation(s) and 32 variable(s).
`<dyn:connect id1="load" var1="switchOffSignal1.value" id2="transformer" var2="switchOffSignal1.value"/>`

So the fact to add .value changes the way equations are counted. I believe this might be due to the way we deal with boolean variables in the extvar files, with the defaultValue attribute, but it is just a guess... Is this behavior normal?  @adriguir @rosiereflo

adriguir, 2020-06-15:
I don't know what's going on. Needs further investigation I suppose, except if @rosiereflo knows by heart ;)

rosiereflo, 2020-06-15:
nope ;) Let me take a look

rosiereflo, 2020-06-15:
Below are my observations.
Without any change we add the following equations to be able to compile:
when time > 999999.0 then
    load.switchOffSignal1.value = false;
    transformer.switchOffSignal1.value = false;

with the following connection:
<dyn:connect id1="load" var1="switchOffSignal1" id2="transformer" var2="switchOffSignal1"/>
Only the connection is added, thus resulting in a missing equation:
load.switchOffSignal1.value = transformer.switchOffSignal1.value;

with the following connection:
<dyn:connect id1="load" var1="switchOffSignal1.value" id2="transformer" var2="switchOffSignal1.value"/>
Both equations are added, thus resulting in a additional equation:
when time > 999999.0 then
    load.switchOffSignal1.value = false;
    transformer.switchOffSignal1.value = false;
load.switchOffSignal1.value = transformer.switchOffSignal1.value;

This is thus caused by the way we handle discrete variables from external vars file. What would be the expected equations?

MarianneSaugierRTE, 2020-06-15:
I would expect:

load.switchOffSignal1.value = transformer.switchOffSignal1.value; (connected to overwritting load.switchOffSignal1.value = false;)

transformer.switchOffSignal1.value = false; (from extvar file because not connected)

@adriguir do you agree?

adriguir, 2020-06-15:
Accidentally closed ;)

adriguir, 2020-06-15:
@MarianneSaugierRTE I finally think you are right. I hesitated quite a lot but I believe this is the right way of doing it. And this is consistent with what is done for continuous variables: for example, efd is only present once in the extvar files (even if being in two models), the same goes for Cm.

Indeed, what extvar file indicates is that you miss one equation for your variable. So if you have A from model 1 missing one equation, and B from model 2 missing one equation, when you connect them you still miss one equation overall. 

The remaining question is which one should stay free to be connected into the dyd file (A or B) -> you can't choose. So we should have a convention that for example the variable remaining free is the first one of the connect statement in the xml. We need to know this in order to know which variable to connect in the dyd file later on.

rosiereflo, 2020-06-18:
I just pushed the branch 874_fix_extvar_file_generation which seems to be working fine when I try different combination. @MarianneSaugierRTE , could you please try it and see if you are able to finish your development with it?

MarianneSaugierRTE, 2020-06-18:
Of course, thanks! I keep you posted

MarianneSaugierRTE, 2020-06-23:
With this branch I can connect load switchOffSignal1 and transformer switchOffSignal1 in LoadOneTransformer and the behavior is fine.
Unfortunately we also have to connect the tap changer switchOffSignal1 for LoadOneTransformerTapChanger, othervise the tapChanger keeps changing the tap because he monitores a null voltage. Here are the new errors:

If I write this I get the error: external variable id (load.switchOffSignal1.value) is not unique
`    <dyn:connect id1="load" var1="switchOffSignal1.value" id2="transformer" var2="switchOffSignal1.value"/>
    <dyn:connect id1="load" var1="switchOffSignal1.value" id2="tapChanger" var2="switchOffSignal1.value"/>`

If I write this I get the error: The model has 49 equation(s) and 48 variable(s)
`    <dyn:connect id1="load" var1="switchOffSignal1.value" id2="transformer" var2="switchOffSignal1.value"/>
    <dyn:connect id1="transformer" var1="switchOffSignal1.value" id2="tapChanger" var2="switchOffSignal1.value"/>`

rosiereflo, 2020-06-24:
I have pushed a new branch with a fix for this. Could you please try it? Thanks!

MarianneSaugierRTE, 2020-06-24:
It works fine with the .value but not without (missing equation) (so it is enough for the connection of switch-off signals but the behavior with/without .value is still different)

MarianneSaugierRTE, 2020-06-25:
Thanks to the switchOff signals connection, we save one discrete variable for each LoadOneTransformer and two discrete variables for each LoadOneTransformerTapChanger :)

adriguir, 2020-06-25:
This is good news and I suppose we can go up to four in the case of LoadTwoTransformersTapChangers

rosiereflo, 2020-06-26:
I fixed the issue, with the final version of the branch we can hopefully  connect n-points with full or partial naming.

MarianneSaugierRTE, 2020-06-29:
In fact, to do things correctly, we would have to have the same number of switchOff signals for the loads and the transformers/tapChangers. Because if the load can be disconnect for the node and from an external event, it is the same for the transformers/tapChangers connected to it. This would require to increment the number of switchOff signals for transformer and tapChanger. If we do this we don't have any gain on the number of discrete variables anymore :( @adriguir  do you agree?

MarianneSaugierRTE, 2020-06-29:
@rosiereflo I pushed a commit on the branch with the connections between switchOff signals for load and transformer/tapChangers. I still have a compilation problem with model LoadTwoTransformersTapChangers. Can you have a look when you'll have a moment?

rosiereflo, 2020-06-29:
sure thing!

MarianneSaugierRTE, 2020-06-29:
Thanks!
And I made a mistake earlier: we still have a gain of discrete variables, even with the increase of switchOff signals number for transformers.

rosiereflo, 2020-06-30:
I pushed a new version. Seems like you are going to trigger all the bugs in the corner cases of the algorithm (which is good ;))

rosiereflo, 2020-07-16:
We want this to go into the release right?

MarianneSaugierRTE, 2020-07-16:
Yes because it should improve the performances for big tests cases with a lot of loads ;) We are waiting for your stats!!

closed 2020-07-16 by rosiereflo