dynawo/dynawo #876: Improve checkDataCoherence method

adriguir, 2020-05-12:
Two ways of improvements in terms of performances could be envisioned:

- As it deals with assert, only call this method in Debug mode (if it is not already the case).

- Separate (if we choose not to do the aforementioned point) the check on parameters that can be done only once at the beginning of the simulation from check on variables values (for example the number of switch off variables is a parameter and is checked at each time step iteration if I am right).

It is not top priority but anyway, @rosiereflo what do you think?

rosiereflo, 2020-05-12:
I prefer solution 2 as  if we skip those checks we might obtain chaotic behavior afterwards. I can add a "checkParameterCoherence" method which would be called after local initialization if your assumption is right

adriguir, 2020-05-12:
Fine for me.
I would just like to get rid of these parameters' check if they really exist (anywhere else that in my mind ;)

closed 2020-05-18 by rosiereflo