dynawo/dynawo #895: Introduce different g types

adriguir, 2020-05-14:
There is currently only one g method, calculating the g values for all root methods, whatever its type and whatever are the depending variables. It could be interesting to distinguish between g roots depending on variable values and g values depending only on time. This distinction could be used to accelerate the computation time with the simplified solver, depending on the system state.

The development will consist in:

- creating the structure with the different g methods

- being able to generate them from both Modelica and C++ models

- utilizing them efficiently on the solver side

rosiereflo, 2020-12-14:
on the solver side we want to :
-compute those G only the first time in the discrete variable propagation loop
- do not recompute the time-dependent G during an algebraic restoration
Do you see any other application?

adriguir, 2022-01-06:
The need for such an evolution can be questioned. It certainly exists other ways to improve performances that will be less complex.

closed 2022-01-17 by adriguir