dynawo/dynawo #903: Build and write in the different output files only when necessary

adriguir, 2020-05-15:
As of today, the jobs file enables to export or not a timeline, export or not the constraints, export or not the logs and so on. Except for logs (which have been recently refactorized and improved), the other infos are built even if they won't be exported later on.

The development targets at avoiding to construct and fill the objects if they won't be exported later on.

rosiereflo, 2020-06-16:
Also as a second step we could evaluate the possibility to dump the timeline and constraints only at the end of a timestep.

adriguir, 2020-10-19:
A few thoughts specific to constraints, where the problem could be a bit different, needing some improvements in the algorithm and structure of Dynawo.

In the current version of the code, constraints are determined, calculated and outwritten in all simulations, whatever is the simulation use case. For some cases, calculating constraints is useless and has a large impact on simulation time (for example in the bus model).

Searching, calculating and determining the constraints should be optional and set by a parameter in the jobs file. It seems quite easy and straightforward to distinguish roots and z equations related to constraints in C++ models but finding an automatic way to do it for Modelica models appear more challenging.

adriguir, 2021-01-19:
Is it still work on progress? I believed it was over for the parts linked to writing in the files if and only if necessary.

rosiereflo, 2021-01-19:
You are right it is over for this part, but updating the G for network buses is not done yet

adriguir, 2021-01-19:
Right, I thought there was another issue for this but I finally see it is included in this one. Sorry ;)!

PR #1374 (2021-01-08): #903 Use timeline events and constraints only if exist
closes #903
closed 2021-03-01 by adriguir