dynawo/dynawo #931: Study the possibility to compute separately state variables used only in root evaluation

rosiereflo, 2020-05-26:
Continuous variables which are only use in roots are added into state variables and associated to 1 equation. This ticket proposes to analyze if we could remove them from the problem and evaluate them only once before the discrete value loop.

rosiereflo, 2020-06-17:
@adriguir This proposal is very marginal: We gain 1 variable per generator and that's it. I propose to put it aside for now.

adriguir, 2020-06-17:
I was always thinking also on this topic on my side. I see a real potential on this question, but certainly in a broader way and in connection with connectors.

When you look to automata, in most of the cases, they have the following structure: they have one to n variables that is external (the controlled variable -  a current or a voltage module) and this variable is connected to a calculated variable from the network. As of today, I think that then you have one to n variables that you don't need to calculate at each f and the same goes for the calculated variable on the network. 

So I believe a new structure / kind of variable would be useful here.

closed 2021-05-28 by rosiereflo