dynawo/dynawo #961: Print a list of higher derivatives at the beginning of a simulation

MarianneSaugierRTE, 2020-06-17:
Maybe a list of 10 to 20 in dynawo.log would be really easy

MarianneSaugierRTE, 2020-06-17:
Maybe a list of 10 to 20 in dynawo.log at first time step would be really easy? @adriguir @rosiereflo @RomainLosseauRTE @gautierbureau

adriguir, 2020-06-18:
From my point of view, it can also be useful for example after events.

rosiereflo, 2020-06-18:
I can propose 2 things; please feel free to add any other idea you might have:
- Dump it in the dynawo log after each step when debug is activated (similar to highest residuals values)
- Add a dedicated log (DERIVATIVES) that would contain the highest values of the derivatives after first time step, zchanges and modes

adriguir, 2020-06-18:
I really have two points that I don't like with option 2: 

-  Derivatives are already dumped (except if we deleted it in the log refactoring, I don't remember) for IDA solver at initialization so this would be only useful for some calculations. I don't really like this.

- I really think that it could be useful at more than one time step.

On the other side, the debug log is quite messy with all the residuals and so on. So I'm afraid it would become unusable. The main problem for me being that it is a flat textual log certainly so you can't merge or unmerge some parts.

Sorry, not a very constructive comment. I will see if I have another idea to propose on my side, otherwise we can do one of yours to go ahead.

rosiereflo, 2020-06-18:
I can also propose to add a verbosity level in the logs to filter which part should be dumped

MarianneSaugierRTE, 2020-06-18:
I don't really know how I could contribute to this discussion but maybe we could start exposing the use we could do with it, and it could help us decide what to do.

As for me, the current need I would have is to have acces to the 10 to 20 highest derivatives after initialization, if possible without compiling in debug mode, and in a log file (dynawo.log seemed a good place to me, as we only talk about a few lines here). 

I also could be interested to have this info after an event, so after algebraic mode restoration (hope the terms are right^^) because it happens that the simulation fails there. But anyway, maybe the re-initialization doesn't succeed in that case, so I don't know if this info will be available... 

I personally won't have a look at this after every z or mode change I think.

rosiereflo, 2020-06-19:
OK so here is my proposal then:  let's dump them in the general log (DEBUG lvlFilter) for now, and if we see that it becomes really difficult to read we might come back on this decision. Is this OK with everyone?

MarianneSaugierRTE, 2020-06-23:
Ok.

For information: On a regional test case with event the dynawo.log file was 665ko and with the development it is 863ko (lvfilter debug)

adriguir, 2020-06-30:
@rosiereflo isn't there a bug in the test in printHighestDerivativesValues()? Here is the code:

if (Trace::logExists("", DEBUG)) return;

I would have expected if !(Trace::logExists("", DEBUG)) return;

rosiereflo, 2020-07-01:
Solved.

closed 2020-07-01 by rosiereflo