dynawo/dynawo #963: Avoid calculating current values at each evalG call

adriguir, 2020-06-18:
As of today, when calling the evalG for current limits, we evaluate the current values, which should be stable once the continuous part has been calculated.

We could save a few evaluations by doing it only once. If #931 is done, the current evaluation should be one of these new variables.

adriguir, 2020-11-09:
As #931 is not planned to be done soon, we can do a proof of concept on this issue and see what could be the associated gains. There is room for at least two potential improvements:

- Directly related to the issue name and current description, we should be able not to recalculate the i1 and i2 variables needed for the current limits root detections (evalG) after a first evaluation at each time step. It means that once it would be calculated by the first evalG at the current time step, the value should be stored in a way or another and only retrieved by the next evalG evaluation. There is one exception to this rule, which is the case of an evolution on the line topology (given by the boolean updateYMat variable at the moment).

- In a similar way, ir and ii which are calculated for each evalF iteration, can be stored in order not to be recalculated in evalCalculatedVars, evalJCalculatedVarI and evalG. It would mean making these objects model attributes and update them at each calculation in evalF before returning them. Note that the calculation still should be done in evalCalculatedVarI because there is no insurance it will be done before or after the line evalF evaluation (at least at first glance): @rosiereflo do you know if the connector calculated variables submodels are placed at a specific location in the submodels vector?

e2l-su, 2020-12-07:
During a sample test (gros recollement) ModelNetwork::evalF() is being called 9958 times, and leads in 6430 cases in calling (*Components)->evalNodeInjector() -AAAA- 
ModelLine::evalNodeInjection() for the same example is called 54 millions of times : 
   - some (83830 times) during isInitModel() phase
   - all other cases as !isDynamic()
   - no cases are met woth other conditions

Optimization should set an indicator saying that in cases mentionned in -AAAA- it is a necessity to calculate i1() and i2(). But for all cases it should be enough to recall the last calculated value.

It is not clear from the little file DYNModelLine.cpp to know who evalG() is called. From Network::evalG(), on the sample test above, the method is called 6974 times. But from traces we see ModelLine::evalG() being called 58 millions of times, which leads to 53 millions calling to i1() and i2() for which we don't want to calcultate all times.

rosiereflo, 2020-12-11:
- The detection of continuous variable modifications has to be done at a higher level than the line as we optimize the calls to evalF, which mean that we would need to modify component API to do the i1, i2 optimizations. I don't think the gain expected worth this major change, so I think we should drop this unless some performance issues linked to this appear.
- regarding the second one we cannot assume the order of the models and for the same reason as above continuous variable might change without evalF of the line being called, so it would need an API update. 
In conclusion I think we should close this ticket, at least for now.

closed 2020-12-11 by rosiereflo