dynawo/dynawo #1063: Improve the Modelica library usability in OpenModelica recent versions

adriguir, 2020-08-05:
There are a few implementation choices in the Dynawo Modelica library that are not consistent with the Modelica language norm and avoid being able to fully use the library in recent OpenModelica versions.

In addition to this, some small features can be added or corrected (graphical views, color and size of some icons)

adriguir, 2020-08-06:
A few other ideas I have in mind, playing a bit with the library to set up a full Modelica test case (RVS Test System, available here https://github.com/dynawo/dynawo/tree/1062_RVSTestSystem) to see if what we recommend to other people makes sense ;):

- The current switch off structure is very annoying when you do full Modelica examples. I would propose not to have the Modelica models to directly extend the SwitchOff model but to add a switch off block/model in the preassembled models.
- It could be nice to have for example different colors for the two terminals of the transformer to know which one is the HV side and which one is the LV side. This could be done by defining two extra connectors called terminalHV and terminalLV, extending the terminal object and only redefining the background color.

@RomainLosseauRTE @cossartque @MarianneSaugierRTE what do you think? do you have other minor improvements to propose?

MarianneSaugierRTE, 2020-08-06:
I totally agree with your first point. Regarding the second point I think it could indeed be helpful to name the terminals HV or LV in order to know which one to connect easily when we are in the diagram view. Nevertheless having different colors for the terminals depending on the voltage level could be confusing: imagine you have a test case with two voltage levels linked by one or many transformers, only the LV transformer pin would have a different color, but all the lines connected to the LV side would have the default terminal color, so graphically speaking the color won't really represent the voltage level and we would end up with only a few pins with a different color. Would it be ok for you?

adriguir, 2020-08-06:
You are right for the second point. My proposal was to have two colors that are different from the other normal terminal color so the terminal colors for HV and LV will be different from the ones from the VL terminals in any case (that's indeed also confusing).

The point is that when you drag and drop a transformer, in the current situation, you have no idea on which terminal is the HV and which terminal is the LV: you need to go and check in the connect statements which is not very practical.
Perhaps having otherwise an indication on the tranformer (like HV and LV written) could help or another idea not using the terminals not to have confusion with the rest would help.

MarianneSaugierRTE, 2020-08-06:
If we rename the pins terminalHV and terminalLV, it will be possible to see which one to connect from the diagram view by letting the mouse on the terminal a few seconds to check whether it is the HV or the LV. Would it be convenient enough?

MarianneSaugierRTE, 2020-08-06:
> g the mouse on the terminal a few seconds to check whether it is the HV or the LV. Would it be convenient enou

[image]

adriguir, 2020-08-06:
This is a step in the good direction. We can try this way and see when we use it if it is convenient enough or not.

What is also a pity is that you can't modify the existing colors when you extend a model. I wanted to do an EHV bus in my example, extending the normal bus and just modifying the color of the rectangle but it is not possible to do this.

adriguir, 2020-08-06:
I managed to cheat a bit by extending the bus object and adding a second rectangle over the first one ;) it works well but is on the border line.

[image]

adriguir, 2020-08-11:
> I totally agree with your first point. Regarding the second point I think it could indeed be helpful to name the terminals HV or LV in order to know which one to connect easily when we are in the [...]

The problem with the switch off signals is that if we output then, we still have the running variable to define locally and so we will still have to add additional equations like lineA.running = true for a line.
Considering this, I finally think that as long as we don't have a solution that will avoid writing additional equations in the final Modelica models, then there is no hurry to change.

MarianneSaugierRTE, 2020-08-12:
> The problem with the switch off signals is that if we output then, we still have the running variable to define locally and so we will still have to add additional equations like lineA.running = [...]

Ok you're right +1:

closed 2020-10-01 by adriguir