dynawo/dynawo #1077: Missing TimelineKeys Class

SoerenLohr, 2020-08-13:
While trying to validate a test model, which contained lines from Dynawo.Electrical.Lines, I encountered the following error:
"[Dynawo.Electrical.Controls.Basics.SwitchOff: 174:7-174:49]: Variable TimelineKeys.LineOpen not found in scope Dynawo.Electrical.Lines.Line$line1." 

Upon further investigation, I found out that there is no TimelineKeys class in the latest version of Dynawo. However, it seems like a lot, if not all, of the SwitchOff classes use TimelineKeys to log events. This prevents the usage of the Electrical models, since the created model can't be flattened/compiled due to missing classes.

Since I don't exactly know what is planned for the SwitchOff Logic, I propose adding the TimelineKeys class in again for the time being to make the models usable and then update the SwitchOff Logic in its entirety.

martinscripts, 2024-03-07:
Hey dynawo team! 

I am repoening this because I am currently [writing with the developers of OMPython](https://github.com/OpenModelica/OMPython/issues/199) to solve a problem. 
I was asked

> Can you please open a [ticket on the dynawo project](https://github.com/dynawo/dynawo/issues) where you ask the developers to remove the timelineKeys files from .gitignore and manage them properly on the git repository? We will soon add Dynawo to the OMC package manager, and this is essential for the whole thing to work properly in all cases. Thanks!

The missing timelinekeys are a problem for usability by others. 
Let's find a solution for this. What is the reason for the files being included in .gitignore?

casella, 2024-03-07:
I'll discuss this with @joyelfeghali and try to find a good solution.

casella, 2024-03-07:
@rosiereflo can you please post a pointer to that?

rosiereflo, 2024-03-07:
Sure!
https://github.com/dynawo/dynawo/releases/download/v1.5.0/Dynawo_Modelica_Library_v1.5.0.zip

joyelfeghali, 2024-03-07:
Hello, I opened a new issue #3158 to treat this problem. I will close this one.

closed 2024-03-07 by joyelfeghali