dynawo/dynawo #1104: Fix full OM test cases

MarianneSaugierRTE, 2020-09-04:
Current error are:
For the SMIBStepPm:  "Illegal access of protected element lambdafPu".
For the HVDC test case and the associated HvdcVSCNoBlockingFunction.mo model in the BaseModel directory: "Illegal access of protected element PInjPu".

MarianneSaugierRTE, 2020-09-23:
Errors related to protected variables could be easily fixed but then we get numerical errors when trying to simulate the Hvdc test case

MarianneSaugierRTE, 2020-12-01:
@cossartque @RomainLosseauRTE @gautierbureau Could you please check the three full OM test cases and tell me which work fine, which doesn't (with error messages), and which OM version you are using (@gautierbureau you can run it which the command line you use in you automatic tool chain). I have different OM environments and none of them can simulate the three test cases properly...Thanks a lot :)

cossartque, 2020-12-02:
On my side : 
- The SMIB case doesn't work because of an "Illegal access of protected element lambdafPu.". Otherwise it works well.
- The grid forming case works well.
- The HVDC case doesn't work for numerical reasons : "kinsols runs into issues retry with different configuration." and "Move forward with a less accurate solution.".
I am using OMCompiler v1.14.0-dev.26789+g1c369fe1a and OMSimulator v2.1.0-dev-147-g36ec2c7-linux.

RomainLosseauRTE, 2020-12-02:
Same results as Quentin's
I am using OMCompiler v1.16.0-dev.578+g4fb5363aba and OMSimulator v2.1.0-dev-177-g0d19b0d-linux

MarianneSaugierRTE, 2020-12-02:
Thanks a lot!

adriguir, 2020-12-09:
@MarianneSaugierRTE is it fine at the moment?

MarianneSaugierRTE, 2020-12-09:
> @MarianneSaugierRTE is it fine at the moment?

No not yet

adriguir, 2020-12-09:
OK thanks, nothing urgent. Just wanted to know

closed 2021-03-25 by cossartque

MarianneSaugierRTE, 2021-03-25:
HVDC test case still gives a numerical error see #1538