dynawo/dynawo #1211: Error messages are not user friendly

mathbagu, 2020-10-16:
I tried to generate a DYD file based on an existing test case ([IEEE57 Disconnect Group](https://github.com/dynawo/dynawo/tree/master/nrt/data/IEEE57/IEEE57_BasicTestCases/IEEE57_2_DisconnectGroup)).

If I run dynawo on my modified test case, the execution fails with an error I can understand (PIN not found) but it doesn't help me to fix my inputs:
```
2020-10-16 18:23:44 | ERROR | subModel NETWORK (type NETWORK) : There is no element _GEN___12_SM_switchOff in the mapElement of the model ( DYNSubModel.cpp:261 )
```

The error message says that the element is not found in a `mapElement` so it should be possible to list the available elements I have access to?

adriguir, 2020-10-17:
In principle I agree with your comment, it would be nice to provide the possible elements. 

From a practical point of view, with for example the network model, the number of elements in the map could be a few thousands for a large scale system. We need to see if there is a nice way to output the possible "structure" in a way or another.

rosiereflo, 2020-10-19:
This error is probably caused by a bad naming of a connection. Could you please send use the files so that we can find out what is precisely the issue and check if we can improve the error message?

rosiereflo, 2020-10-19:
@adriguir we could maybe output only the possibilities for the blackbox model with a factorization of the pins with an index? For example for VariationArea: deltaPc_load_<0-1000>?

adriguir, 2020-10-19:
> @adriguir we could maybe output only the possibilities for the blackbox model with a factorization of the pins with an index? For example for VariationArea: deltaPc_load_<0-1000>?
> Moreover I think the term Element is misleading here, we should at least write clearly that this signal does not exist

For "vectoral" structures as in VariationArea or OmegaRef, this would be perfect. For network, perhaps we can have something like <id>_ACPIN for example? Even if you still have to retrieve the id yourself, it really has an added value by giving you the generic syntax structure (especially is there a _value?) which is always problematic.

So we can try to achieve such a level of indications in the log.

mathbagu, 2020-10-19:
Yes this is my diagnostic too. But could you explain how do you convert a `dyn:connect` to `dyn:macroConnect`.
```
<dyn:connect id1="DisconnectGenerator" var1="event_state1" id2="GEN___12_SM" var2="generator_switchOffSignal2"/>
```

I tried this:
```
    <dyn:macroConnector id="MC_EventSetPointBoolean">
        <dyn:connect var1="event_state1_value" var2="@NAME@_generator_switchOffSignal2"/>
    </dyn:macroConnector>
```

rosiereflo, 2020-10-19:
Proper macro connect for this connection:

  <dyn:macroConnect connector="MC_EventSetPointBoolean" id1="DisconnectGenerator" index1="0" id2="GEN___12_SM"/>
  <dyn:macroConnector id="MC_EventSetPointBoolean">
        <dyn:connect var1="event_state1" var2="generator_switchOffSignal2"/>
    </dyn:macroConnector>

mathbagu, 2020-10-19:
Could you explain why for the `EventQuadripoleDisconnection` I have to write `@name@_state_value`? I don't understand to what the `@name@` refers to...

rosiereflo, 2020-10-19:
the @name@ is the name given as id in the maccroConnect. In the network, we need to prefix the signal name by the object name to identify it properly (as the network is a  container of all objects with no dynamic model defined): 
  <dyn:connect id1="DISCONNECT_LINE" var1="event_state1_value" id2="NETWORK" var2="**_BUS____1-BUS____5-1_**AC_state_value"/>
Regarding the macroConnect above, both models are blackboxes and thus there is no need to prefix with the name

closed 2021-01-08 by adriguir