dynawo/dynawo #1212: Constraints search and output

adriguir, 2020-10-17:
In the current version of the code, constraints are determined, calculated and outwritten in all simulations, whatever is the simulation use case. For some cases, calculating constraints is useless and has a large impact on simulation time (for example in the bus model).

Searching, calculating and determining the constraints should be optional and set by a parameter in the jobs file. It seems quite easy and straightforward to distinguish roots and z equations related to constraints in C++ models but finding an automatic way to do it for Modelica models appear more challenging.

rosiereflo, 2020-10-19:
isn't this a duplicate of #903 ?

adriguir, 2020-10-19:
You are right, I haven't noticed that we already had a generic issue around outputs open, my fault.

closed 2020-10-19 by adriguir