dynawo/dynawo #1235: Delay handling for vector structures

adriguir, 2020-11-01:
As of today, delays defined in a vectorized form in Modelica are not supported in a correct manner. The regex test to detect delay maximum values as well as the variable associated to the delay is failing.

Below is an example of a _7dly.c file that doesn't work properly:

storeDelayedExpression(data, threadData, 0, data->localData[0]->realVars[40] /* Tlm.historyTerm.sVm[1] variable */, data->localData[0]->timeValue, data->simulationInfo->realParameter[67] /* Tlm.historyTerm.tau[1] PARAM */, data->simulationInfo->realParameter[67] /* Tlm.historyTerm.tau[1] PARAM */);

lecourtoisflo, 2020-11-03:
can you provide a set of inputs that generates this error

adriguir, 2020-11-03:
For sure, you can checkout the 000_OpenEMTLibrary and unset the horrible modifications I have done in the readerOMC.py to handle my case (I'm really not at ease with regex so wasn't able to correct it myself).
The preassembled model that has a problem is the TLM model.

PR #1238 (2020-11-03): #1235 Correct parsing of delays
fixes #1235
closed 2020-11-03 by adriguir