dynawo/dynawo #1236: Corrections following some solver debug on testcases

gautierbureau, 2020-11-02:
Those issues were encountered when I tried to debug the resolution of some test cases

rosiereflo, 2020-11-02:
Please be more specific  so that if someone encounters the issue he can retrieve this ticket

gautierbureau, 2020-11-02:
I can't. People won't get into those issues as they will be corrected with the PR

rosiereflo, 2020-11-02:
My opinion is that we should be as specific as possible in the tickets to have a proper history of fixes. A few words will be enough (e.g. some debugger optimizations are enabled when models are compiled in Debug, preventing the proper use of debug tool for example)

gautierbureau, 2020-11-02:
I agree but it was 3 different issues I encountered a while back and never corrected them and don't really remember what was the issue. The -O0 is indeed to fully debug models but I forget it was in it... The cmakelists is just a better handling of transitive dependency and the last one is a comment on something I always forget and have to look up for in the sundials doc to know how to do exact newtons instead of inexact ones

closed 2020-11-04 by rosiereflo