dynawo/dynawo #1248: Change conventions of P and Q init values for DynaFlow HVDC models

cossartque, 2020-11-09:


adriguir, 2020-11-09:
Would it change the conventions compared to other models? Is it due to a difference in the IIDM? 
Please also double check with @MarianneSaugierRTE to keep it quite consistent in the whole library if the change is done in the model and not in the interface.

cossartque, 2020-11-09:
For the moment, the dataInterface of the converters has a different convention compared to the other interfaces (load, generators).
I would prefer to change it in the dataInterface but I suppose it was done this way for a reason. What do you think @MarianneSaugierRTE ?
If we can change the interface, then no change is needed in the DynaFlow models.

cossartque, 2020-11-10:
I just realized that instead of using the dataInterface of the converter, I can use the data interface of the hvdcline, which uses the convention that I need. I will close the issue.

closed 2020-11-10 by cossartque