dynawo/dynawo #1256: Optimize J evaluations for Modelica models

adriguir, 2020-11-12:
In the current implementation, the Jacobian matrix for the Modelica models is derived using an automatic differentiation method (Adept). It notably implies redoing an evaluation of the residuals using adept types. Possible improvements could be:

- To avoid useless treatments in the specific residual evaluations using adept types. Constant are for example not necessary for the J calculation and could be deleted from the generated code. 

- To reuse constant parts of the J when the expressions are order 1 and constant all over the simulation. For example, when we have something like k * x for example.

- To investigate the possibilities offered by OpenModelica on J structures

rosiereflo, 2020-11-13:
Hi,
The first and second point would require a syntactic analysis of the computation of the f to determine where are the constants, if they are not included in a multiplication and so on, which would be complex and probably buggy (case where there is a if for example...). I always tried to avoid this and I would prefer to rely on OM as they already do those kind of analysis to generate the code.

adriguir, 2020-11-13:
Open ideas to be discussed, no developments going on.
The second point is clearly inspired by recent proof of concept by Braun and Bachman in this direction - https://www.researchgate.net/publication/345677824_Contributions_to_the_Efficient_and_Parallel_Jacobian_Evaluation_and_its_Application_in_OpenModelica, unfortunately not yet available in OM if I get it right.

adriguir, 2020-11-26:
Closing it at the moment as no really long-term viable option without OM improvements/developments.

closed 2020-11-26 by adriguir