dynawo/dynawo #1257: Complete the DynaFlow Hvdc library

cossartque, 2020-11-16:
This issue aims at adding more models to the DynaFlow Hvdc library:
- Add a preassembled model for HvdcPV with a SetPoint for URef1Pu and URef2Pu (for the moment there is only a preassembled model for HvdcPVEmulation);
- Modify the preassembled model of HvdcPTanPhi to add a SetPoint for TanPhi1Ref and TanPhi2Ref;
- Add an HvdcPVQ model and the associated preassembled models (HvdcPVDiagramPQ and HvdcPVDiagramPQEmulation). With this model, terminal1 regulates the voltage while terminal2 regulates the reactive power;
- Add an HvdcPVDiagramPQ model and the associated preassembled models (HvdcPVDiagramPQ and HvdcPVDiagramPQEmulation);
- Add an HvdcPVQDiagramPQ model and the associated preassembled models (HvdcPVQDiagramPQ and HvdcPVQDiagramPQEmulation);
(- Add an HvdcPVDiagramPQDangling model and the associated preassembled model (HvdcPVDiagramPQDangling).)
@marcochiaramello and @adriguir , any idea/comment/suggestion on this is welcome !

adriguir, 2020-11-16:
Thank you for the proposal.
One remark that comes to my mind (but I have not really thought about it) is on the way to organize things: is it better and nicer to have separate models with different regulation frames or would it be better to have less models with different regulation possibilities (like Q or V regulation mode)? 
For example, in the HVDCVSC model, the model can handle either Q or V regulation if I remember correctly.
@MarianneSaugierRTE any opinion on this?

MarianneSaugierRTE, 2020-11-16:
Regarding V/Q regulation, as we can think that there will always be a parameter somewhere to give the regulation mode (we have it in iidm, I guess it is also in CGMES?), it could be possible to have it in the model and use it to active the good behavior, and have only one model. This could also be done for the active regulation mode: setpoint/emulation but I am not sure it is as common as for the V/Q mode. 

For other characteristics I would suggest to stick with different preassembled models as Quentin propose, for example with or without diagrams, etc...

Hope you could understand what I meant ;)

cossartque, 2020-11-16:
In my opinion, the information that we get from the IIDM helps us to apply the proper preassembled model in the .dyd (PTanPhi if lcc, PV if vsc with two V controls, PVQ if vsc with one V control and one Q control), with or without AC emulation, dangling if needed, etc. This would be done by the launcher.
In the library, as we have a separation between PTanPhi and PV, I would suggest to have a third model PVQ instead of an option V/Q that would be applied to the PV model.
However, I suggest that we can create several base classes models that we use in the three types of general .mo models (PTanPhi, PV, PVQ).
What do you think ?

adriguir, 2020-11-16:
I think you line of thinking is quite valid if we always imagine using the launcher. It will be the case for all RTE-based uses but we can't prevent someone else using DynaFlow directly without using the launcher. That is a first point (not very important).

In addition, I clearly see for example a distinction between for HVDCPTanPhi and HVDCPV because it's lcc or vsc and one component can't switch from one simulation to another from HVDCPTanPhi to HVDCPV and vice versa. This kind of perimeter of model seems fine in my opinion. However, for PVQ or PV, we can imagine a component being HVDCPV or HVDCPVQ depending on the simulation (or the horodate) so this is a bit more unclear. And perhaps one day some simulations could consist in changing on the fly from Q to V regulation or vice versa.

Of course, it's all philosophy questions and each option has its pros and cons. But I would tend to deal with the V/Q choice in one model. When implementing, if it turns out to be too uggly, we could then change our mind.

cossartque, 2020-11-17:
Thank you both for your answers. Then I will try a V/Q implementation :)

closed 2020-11-23 by adriguir