dynawo/dynawo #1272: Make Modelica Library compatible with last OM version

MarianneSaugierRTE, 2020-11-24:
1. lookup (variables and parameters should be used where they are declared)
ex: Variable s0Pu not found in scope BaseLoadInterfaceParameters_INIT.

2. protected (we sometimes access protected elements illegaly)
ex: Illegal access of protected element u0Pu.

3. invalid connections  (we sometimes use variables inside connects)
ex: uPu.re is not a valid connector, PmPu.value is not a valid connector

4. variables discrètes et ZPin
ex: Following variable is discrete, but does not appear on the LHS of a when-statement: ‘tap.value‘

MarianneSaugierRTE, 2020-11-24:
4. No correction to be made: the model can still be simulated when ZPin are connected in a full/global model

adriguir, 2020-11-24:
Should also resolve #1104 I believe.

PR #3589 (2025-03-17): #1272 first adaption of Modelica library for OpenModelica v1.24.5 (connectors)
closes #1272

**Checklist before requesting a review**

*use `'[x]'` to check the checkboxes, or submit the PR and then click the checkboxes*

- [ ] unit tests and non-regression tests were added (new model, new feature and bug fix)
- [ ] main documentation was updated (update of input/output file, 3rd party, model, repository organization, solver)
- [ ] example documentations were updated (new example in examples folder)
- [ ] the corresponding milestone was added in the ticket and in this PR
- [ ] if this PR modif [... cut]
PR #3601 (2025-03-18): #1272 Miscellanous fixes forgotten by previous commits
closes #1272

**Checklist before requesting a review**

*use `'[x]'` to check the checkboxes, or submit the PR and then click the checkboxes*

- [ ] unit tests and non-regression tests were added (new model, new feature and bug fix)
- [ ] main documentation was updated (update of input/output file, 3rd party, model, repository organization, solver)
- [ ] example documentations were updated (new example in examples folder)
- [ ] the corresponding milestone was added in the ticket and in this PR
- [ ] if this PR modifies the parameters or in [... cut]
commit 790ef595 (2025-03-18): #1272 Miscellanous fixes forgotten by previous commits
closes #1272
commit 6000dc41 (2025-03-25): #1272 Miscellanous fixes forgotten by previous commits
closes #1272
closed 2025-03-17 by joyelfeghali