dynawo/dynawo #1311: Extend the CPP restorative load component to add voltage limitations

adriguir, 2020-12-03:
In its current state, the restorative load component available in the network model doesn't include any voltage limitations: it means that the restoration is done whatever is the voltage value.
A more realistic model will also implement and use minimal and maximal voltage limitations. Outside these values, the load will then behave as a pure alpha-beta load.

Even if this model's improvement is more realistic, it should remain optional for applications not wanting to use it at first stage (DynaWaltz for example). It means that the values should remain optional in the parameters set and the associated g and z shouldn't be set and evaluated in case the limits are defined.

adriguir, 2020-12-03:
We can imagine different implementations for this: having an additional generic parameter in the network set to set if limits have to be enforced, having only the max and min values as optional and use this to set the nb of z and g dynamically or any other idea. 

Any opinion @rosiereflo @cossartque @MarianneSaugierRTE @gautierbureau ?

rosiereflo, 2020-12-04:
I like the second idea: optional min and max per load.

cossartque, 2020-12-04:
Same as @rosiereflo, I think it is better to have only optional min and max.

marcochiaramello, 2020-12-04:
same as @rosiereflo too

adriguir, 2021-01-19:
Update on this issue: it seems the equations written for the long-term voltage stability and for the steady-state problem will differ. A specific load model will certainly be needed, either as a new possibility in the current load model or as a new proper model.

PR #1455 (2021-02-24): #1311 adding cpp model load restorative with limits
Closes #1311
closed 2021-02-24 by rosiereflo