dynawo/dynawo #1413: Avoid some copies at the solver level for the simplified solver

adriguir, 2021-01-26:
It seems a few copies can be avoided in the simplified solver current implementation. 

The initial values of variables y is both stored at the simplified solver level through the ySave_ vector and at the KINSOL level through the y0_ vector. Only one save can be kept and the other deleted. As y0_ is needed anyway for the differential variables evaluation, it seems natural to delete the ySave_ from the simplified solver. In case of non convergence, y0_ can be used to reinitialize the problem.
It also means that the saveContinuousValues/restoreContinuousValues methods are no longer needed.

The setInitialValues for the KINSOL solver can also be simplified. Indeed, doing a zero-order prediction, y0 and vYy_ when beginning a new time step should be equal and so we can simply copy vYy in y0.

It doesn't seem necessary to retrieve the final values obtained by KINSOL into the simplified solver structures at each time step as these values won't be used (at least at first glance), except if we have to do an algebraic restoration.

adriguir, 2021-01-27:
Finally, it also seems that a g copy can be avoided before calling the evalZMode loop  as the final g will be copied in the evalZMode method, whatever happens there.

adriguir, 2021-01-27:
Needs #1255 to be finalized and integrated.

adriguir, 2021-02-08:
Only a part of it is currently done and handled by #1427 .
The others should be discussed because they can make the reading and understanding of the simplified solver implementation more difficult, and quite specific to the current approach.

By order of easyness and understanding by an external view, we can:

- Avoid saving the values of vYy in SolverSIM structure before the time step as they are also saved into DYNSolverKINEuler y0 structure. In case of divergence, we can just use these values for initializing the vYy structure of DYNSolverKINEuler. It saves one copy of y by time step iteration and is quite easy to implement and understand.
- Modify the behavior of the setInitialValues method. Indeed, as we do order-0 prediction at the moment, the y0 structure of DYNSolverKINEuler can be initialized by vYy_ of the DYNSolverKINEuler structure, instead of copying the vYy_ from SolverSIM to the SolverKINEuler values. It works well when the status is convergence and there is no reinit(). In case of reinit, the current scheme should be applied. It saves one copy in most of the cases.
- Modify the behavior of the getValues method. Indeed, in most of the cases, there is no need to retrieve the values in SolverSIM structure and they can be outdated. The only case where we will need vYy_ to be updated is when we go in a reinit(). At this stage, it would be possible to retrieve it from the KINEuler structure. It saves two copies (vYy and vYp) at most of the time steps (98 to 99%). The dangerous point being that we have a vYy and vYp structure in the solver that is outdated.

IMHO, the first point can be done without endangering the long-term maintenance. Points 2 & 3 are open to discussions. I can make a PoC to help seeing the new implementation we will end up with. 

@rosiereflo @gautierbureau  what do you think?

gautierbureau, 2021-03-22:
I agree that copies could be avoided. For me there should be a clear separation between SIM and EulerKIN: first one handles the time-scheme (here dy/dt = (y_n+1 - y_n ) / dt) and the other one should handle the newton resolution. This means KINEuler should only have one vector (F_) and others should be reference to the one managed by SIM. In fact YP_ and y0_ are time scheme related concepts and should not appear in a Newton resolution.

closed 2021-06-07 by gautierbureau