dynawo/dynawo #1445: Add info on frequency value in the log when Dynawo crashes

MarianneSaugierRTE, 2021-02-15:


adriguir, 2021-02-17:
A proposal would be to have both a log and a throw if the frequency becomes lower than a certain value (to be defined). The throw could be catched by the calling method and used to output from the simulation with a specific status (frequency collapse for example).

@rosiereflo I suppose it could easily be done by implementing a checkDataCoherence method in the OmegaRef model, checking the value of omega.

rosiereflo, 2021-02-17:
yes, today the method checkDataCoherence is overriden (and empty) in DYNModelCPPImpl today but it would make sense to add a OmegaRef specific implementation with this check.

adriguir, 2021-02-17:
BTW, if we want to have a bit of flexibility, we can say that the threshold for frequency collapse is a parameter, read in the par file.

rosiereflo, 2021-02-17:
> BTW, if we want to have a bit of flexibility, we can say that the threshold for frequency collapse is a parameter, read in the par file.

I totally agree, hard coded threshold are never good

adriguir, 2021-03-04:
The method checkDataCoherence can be transformed to virtual and not pure virtual and a specific error type is perhaps needed.

closed 2021-03-25 by rosiereflo