dynawo/dynawo #1525: Entso-E and VSC models simulation problems

LlucFigueras, 2021-03-18:
Report comparing Modelica and Simulink models, according to ENTSO-E_131127_Controller_Test_Report, the three test cases plus one extra test case.

VSC model problem commented on the same report, part 2.

[ENTSOE_Report_Comparison_Dynawo_Simulink.pdf](https://github.com/dynawo/dynawo/files/6165632/ENTSOE_Report_Comparison_Dynawo_Simulink.pdf)

adriguir, 2021-03-18:
@LlucFigueras thanks for opening the issue and for the branch. We will try to have a look to this as soon as possible and let you know! 
@cossartque @MarianneSaugierRTE is the results obtained by Lluc and CITCEA in their report is similar to the ones you got in a full Modelica approach for the ENTSO-E test report?

MarianneSaugierRTE, 2021-03-18:
We observe very similar results in comparison with Eurostag, as shown in the following report (only theta is different because the initial angle is a bit different). The simulation tagged "Dynawo" are made using Dynawo models but are actually obtained with OpenModelica. For your information, the models of our Dynawo library are used for the electrical part and the regulations have been implemented in a specific package, not yet open-source. 
[Validation_ENTSOE.pdf](https://github.com/dynawo/dynawo/files/6166250/Validation_ENTSOE.pdf)

If needed, we can share the csv files containing the simulation outputs in order to compare them to what you get on your side.

rosiereflo, 2021-03-19:
Hi, thank you for the branch. Could you please associate each issue with one of the testcase that you added so that I take a look at the correct example? Thanks!

rosiereflo, 2021-03-19:
In particular I am not able to reproduce the "variable number 72 not equals to the equations number 62" error

LlucFigueras, 2021-03-19:
I have opened a new issue, #1527, regarding only the VSC, and updated the vsc model in CITCEA branch. I needed to modify the model because previously I had the models in my database, not in the vsc folder. Now I have been able to reproduce the error "variable number 66 not equals to the equations number 56" error.
Thank you!

LlucFigueras, 2021-05-13:
The first tests VSC simulation is executed successfully. It doesn't crash when the power reference is changed. Thanks.

closed 2021-06-02 by rosiereflo