dynawo/dynawo #1661: Modify the way preassembled models work

adriguir, 2021-05-19:
The current way to do final precompiled blackbox models in Dynawo is by creating preassembled models by xml files that are compiled during the project compilation. 
This approach works fine but is not natural for a model developer or a researcher and makes it more difficult to both spread Dynawo use as well as motivate people to go into Dynawo frame.

One alternative that would deserve further investigation in my opinion would be to switch to a real Modelica-based approach here too by creating in Modelica the final dynamic and init models and writing the connect statements inside Modelica models instead of through xml statements that are finally translated to Modelica models.
One byproduct/byresult would be that we also end up with composite Modelica models that can be used into full Modelica test case, avoiding to do the work twice (BTW, I prefer the word composite model instead of preassembled model).

@rosiereflo @gautierbureau @bongrainmat @cossartque @marcochiaramello what is your opinion? do you already see some bareer for such a change?

cossartque, 2021-05-19:
If I understand it right, this would have an impact only for the preassembled models that connect several Modelica models.
For example, for models like GeneratorPQ.xml or GeneratorPV.xml, nothing would change ?
For more complex models like GeneratorSynchronousFourWindings.xml, it would now only need a GeneratorSynchronousFourWindings.mo and a GeneratorSynchronousFourWindings_INIT.mo that need to be created ?
It seems doable for me but some difficulties could be:
- when the connect is done with connectors and not RealInput/RealOutput, we won't have a graphical view in the newly created .mo models. It is ok for you or do we switch to a full RealInput/RealOutput use ?
- we might need to create intermediate folders in the library to put the models and the composite models on a different level and avoid any confusion
- I suppose some names will need to be changed to avoid confusion too.

adriguir, 2021-05-19:
For the first point, we can also imagine not having anymore xml files but just searching for names in the directory (or having just a list): everything is open - we just brainstorm at the moment.

I see it a bit like this:
Models/Modelica/Dynawo with all the unitary models and then Models/Modelica/CompositeModels which is another package (or not) using Dynawo and potentially other libraries unitary models.
In this CompositeModels, you have structure and you just put somewhere in a file the list of the CompositeModels you want to compile. 

Notice also that I consider changing the way SwitchOff is done which will end up in not having anymore just one model in the xml but at least GeneratorPV + SwitchOff (see #1660 ).

To be further discussed. I can try to create a few examples to have a better basis for discussion.

cossartque, 2021-05-19:
Ok thanks for your answer.

We would also need to put in this list the different associations between .mo and _INIT.mo, no ? Because some _INIT.mo models are used for several preassembled models and therefore don't have the same name as the associated .mo file.

In this list, you would have a list of CompositeModels but also of classical Dynawo models. Some are indeed not composite but "standalone", even when we change the handling of switchOff (the events for instance) and I suppose we won't do a copy (or a simple extend) of them in the new composite library.