dynawo/dynawo #1681: Error when launching job: basic_string::_M_construct null not valid

LlucFigueras, 2021-05-23:
Hi, I obtain the following exception when launching a job: 'Exception: basic_string::_M_construct null not valid'.
The iidm file is obtained using the psse importer in Powsybl. Then, running a load flow, the state variables for generators, loads, transformers and lines are added in the iidm file. Even building Dynawo in 'Debug' type, I don't find where this issue comes from... 
I attach the test case. Thanks.
[IEEE118_Fault.zip](https://github.com/dynawo/dynawo/files/6528937/IEEE118_Fault.zip)

adriguir, 2021-05-25:
Hi @LlucFigueras , the problem seems to come from the shunt entries in the iidm file. The first one works fine because it has the attribute currentSectionCount but the other remaining ones have an attribute "sectionCount" instead of an attribute "currentSectionCount". By replacing the former by the latter, the iidm file seems to be correct and I then go into another issue with the dyd file (apparently one id should be duplicated). 
I will check the id and let you know but it shouldn't be big deal (copy/paste error I suppose)

LlucFigueras, 2021-05-25:
Hi @adriguir, many thanks. The thing is that I couldn't check where the error came from... I have fixed a few other issues and now the simulation executes ok. I will make some modifications in the machines types (for synchronous condensers) and, if you think it is useful, upload it again.

adriguir, 2021-05-25:
No problem. Indeed, there was no real way to find out the problem origin (no further information in the logs file).

I had the intuition that it came from the iidm file because I haven't yet crossed this error since I work on Dynawo and then I searched for unusual components (or at least components that were not present in other test cases) which led me to the shunt component and there I compared with other example cases we have internally in RTE. I was quite lucky that it was the real problem cause... !

Yes please upload it once it is ready. It will be another test case we can add, this is always a good thing! Thanks!

closed 2021-05-26 by rosiereflo