dynawo/dynawo #1700: Event not working at restart with event value set to zero

gautierbureau, 2021-06-01:
In the following example we try to do a variation setPoint on Q of the generator2 at the restart t=10s, in the current behavior the event is not taken into account. It can be seen because generator2_Qc does not go to 0 and no event appears in the Timeline. 

A first analysis hints the issue is that the vector of discrete variables is initialized to zero and then no changes is detected to propagate a change in the connection from the event to the model it is connected to. In `ModelMulti::init(...)` we cans see:

``` cpp
if (doubleNotEquals(zLocal_[i], zSave_[i])) {
  indicesDiff.push_back(i);
  valuesModified.push_back(zLocal_[i]);
}
```

here no changes is detected because both values are equals to zero.

One way to simulate this would be to set the `q` attribute in the iidm to 0 for the restart.

We have to think a bit more what would be the right solution or if we need to simulate this kind of event at restart.

[TestEventNotWorking.zip](https://github.com/dynawo/dynawo/files/6574367/TestEventNotWorking.zip)

gautierbureau, 2021-06-01:
One wrong solution was to change the initial value of the event in `dynawo/sources/Models/Modelica/Dynawo/Electrical/Events/Event.mo` by `typeConnector state1 (value(start = -99999)) "Current state";`

rosiereflo, 2021-06-01:
The loop mentioned is not the right place to do it as it just detecting the modifications of Z during the init of the network. Is the event happening in the first simulation or in the reload simulation?
If it is in the first simulation then there is a bug as the Y0 value should be already 0
It it is in the simulation with the initial state then we would need at some point to call evalZ and to compare with the Y0 value that should be different from 0.

closed 2021-06-10 by adriguir