dynawo/dynawo #1706: IEEE 118 bus system test case

LlucFigueras, 2021-06-02:
Hi,

I attach the IEEE 118 bus system test case. I followed the same structure as IEEE 14/57 DynaSwing example models, the machines data is arbitrary. 
[IEEE118_Fault.zip](https://github.com/dynawo/dynawo/files/6585887/IEEE118_Fault.zip)

adriguir, 2021-06-03:
Hi Lluc,

Thanks for it! 
That's fine for me to add arbitrary dynamic data because I don't think there is any reference and widely accepted ones but perhaps, it's just I don't know them. On your side, how did you proceed when working with the test case in other tools?

LlucFigueras, 2021-06-03:
Hi Adrien,

Glad to help! I proceeded in the following way:

1. Take the IEEE 118 bus system raw file from Powsybl examples.
2. Import the raw file in PSS/E to check the data properly with a GUI, then export the file as raw 33 version. 
3. Using the psse importer in Powsybl, convert the raw file to xiidm.
4. Launch a load-flow using Powsybl, so the load-flow state variables are saved in the xiidm file.
5. Finally, I modified the shunt data format and the xmlns 'tag' (I guess it means xml namespace, but I am not sure..)

When working with the test case in PSS/E, once the raw file is imported, I build the network and add the dynamic data to the machines. Regarding the dynamic reference data, I have searched for it, but I haven't found anything either.

PR #3038 (2023-10-12): #1706 Creating the IEEE118 test cases (adaptation of Figueras' work)
closes #1706
closed 2023-10-12 by joyelfeghali