dynawo/dynawo #1933: Fix computation of lost equipment

zamarrenolm, 2021-10-18:
Equipment completely connected at the initial state and then only partially disconnected at the final state should NOT be considered as lost. Also, a partially connected equipment at the initial state and totally disconnected at the final state should be considered as lost.

Currently, to determine the list of lost equipment, the `isConnected` method in the data interface is used to check initial and final state of an equipment.

The interpretation of `isConnected` for equipment with more than one end states that it is connected only if all the ends are connected. This interpretation must not be changed. With this interpretation for `isConnected`, lines that are initially connected at both ends and are left opened in the final state only at one end are considered as lost. This is considered wrong.

To fix the current implementation of the computation of lost equipment, a new API `isPartiallyConnected` will be proposed.

zamarrenolm, 2021-10-18:
Lost equipment are calculated at the Data Interface level.

First, all connected Data Interface Components are obtained for the initial state of the simulation (`DataInterface::findConnectedComponents`).
After simulation ends, Components that were initially connected and are now disconnected are considered lost (`DataInterface::findLostEquipment`).

Implementations for these methods rely on ComponentInterface::isConnected to determine the state of connection. We propose to add a new method to ComponentInterface, isPartiallyConnected. It must return true if ANY of the ends of equipment is connected.

For equipment with only one terminal (one end, `InjectorInterface` in `PowsyblIIDM`), `isPartiallyConnected` will be equivalent to existing `isConnected`.

gautierbureau, 2022-06-17:
Some unittests were broken and it went silent because the unittests were not run. The PR https://github.com/dynawo/dynawo/pull/2361 had bakc the unittest and we need to make the proprer changes to handle PartiallyConnected lost equipements

PR #1950 (2021-12-15): #1933 Consider partially connected elements in lost equipment report
Closes #1933
commit 53407353 (2022-09-26): #1933 Consider that a partially connected component is not lost
closes #1933
closed 2022-09-26 by gautierbureau