dynawo/dynawo #2019: minimalAcceptableStep should be hardcoded in SIM

gautierbureau, 2021-11-29:


rosiereflo, 2022-01-11:
Why?

gautierbureau, 2022-01-26:
should not*. 

The value here https://github.com/dynawo/dynawo/blob/master/dynawo/sources/Solvers/FixedTimeStep/DYNSolverCommonFixedTimeStep.cpp#L84 is completly arbitrary and should be removed

rosiereflo, 2022-01-26:
ok I get it better now :)

adriguir, 2022-01-27:
I agree, but take care. It is also defined in the constructor in SolverImpl if I'm not mistaken: https://github.com/dynawo/dynawo/blob/fefb60256014f2599ad0279a0664d49cf680691d/dynawo/sources/Solvers/Common/DYNSolverImpl.cpp#L92

rosiereflo, 2022-01-28:
Yes this was added when we couldn't add the parameter on the platform and thus we needed a proper default value in the solver SIM by default (as it needs to be higher than for IDA). This value can be easily overridden by a parameter, and I'm not sure that we want to make this parameter mandatory as it requires some understanding of the solver to chose a proper value, so I propose to let this as it is, or to align it with the value used in the platform currently (which is 1.0).

closed 2022-03-08 by rosiereflo