dynawo/dynawo #2102: MSL & Dynawo : issues with blocks

ErwanGuichard49, 2022-01-19:
Modelica.Blocks.Math.Abs : in Dynawo simulations, this component appears to give the opposite of its supposed output. Problem solved by adding the modifier (generateEvent = true). An equation without the noEvent function is therefore applied : **if u >= 0 then u else -u** instead of **if noEvent(u >=  0) then u else -u**

Modelica.Blocks.Nonlinear.Limiter : from some random point in a simulation, the lower limit is ignored, or the output equals the lower limit while the input is well above this limit. These behaviors were not wiped out with the available modifiers _strict_ and _NoHomotopy_, which indicates that these issues might originate from the smooth function.

Modelica.Blocks.Continuous.SecondOrder : in a model where this component was followed by FixedDelay, the compilation failed. In another one without a delay on the branch, the compilation succeeded. With an equivalent TransferFunction instead of SecondOrder (followed by FixedDelay), the compilation succeeded.

ErwanGuichard49, 2022-01-20:
A workaround has been found : the SecondOrder block has been rebuilt using elementary blocks (integrators, gains, feedbacks) equivalent to the MSL component's equation. It works under Dynawo, its results are identical to those from a TransferFunction block, which can be simulated as soon as its output y is not used elsewhere.

adriguir, 2022-01-24:
Regarding the limiter problem (at least), it seems that the issue doesn't arise in a systematic way. For example, in the GeneratorSynchronousFourWindingsTGov1SexsPss2A preassembled model, the smooth function is called three times: the generated Modelica code doesn't contain the associated relation_hysteresis structure and zero crossing functions but the Dynawo code (*_Dyn.cpp) has the methods created and added, which means the mechanism works in simple cases.

The issue noticed here is then not a systematic issue but should be particular to the system being solved in the test case (needs to have a looked to the generated code).

adriguir, 2022-01-24:
The problem for the limiter and the abs blocks is identical: we create the relation hysteresis structure in both cases but the link between the relation and the testing condition is erroneous in some cases (mixing up things)

adriguir, 2022-01-24:
For the TransferFunction and SecondOrder blocks, the seg fault comes from the following internal OM data structure: data->modelData->realVarsData[*].attribute in the getY0 initialization method.

This data structure originates from the _08bnd.c file in our case, which is certainly not filtered for this kind of structure. 

The Python script should be able to replace the assignment of this value by the parameter and the assignment of the init value by the attribute by a direct assignment of the value by the parameter (skipping the OM internal data structure).

PR #2132 (2022-01-27): #2102 #2109 Fixes compilation scripts and model manager
closes #2102
closes #2109
- Make sure index of created relations is coherent between evalMode and evalF
- properly handle variable initialization
- call customParametersConstructors on init models
closed 2022-01-27 by adriguir