dynawo/dynawo #2124: Use of precision in addTerms in SparseMatrix

adriguir, 2022-01-25:
In the current implementation, in the method to addTerm in the Jacobian matrix, the dynawo precision is used to filter the terms to use (only non zero elements are added).

Nevertheless, in some cases, especially with IDA solver, when we do the algebaic restoration step on derivative variables, we end up with a raw or a column of zero due to this threshold, which then leads to a simulation fail. The countermeasure now consists in decreasing the overall simulation precision, which works but affects all the parts of the code where the precision is used.
It notably appears in WECC test cases and in other test cases using PE based (and advanced controlled structures). Notice that there is no special problem with the simulation if we decrease the precision...

I'm not sure that using this precision at this specific place in the code is the right choice: it seems we need to be more flexible there to be able to simulate more cases.

adriguir, 2022-01-25:
In the past (a long time ago), the value used to consider something equal to zero in this specific method was 1e-12. It is an arbitrary value but it seems to work fine and would make the job in all the test cases I have seen until now.

If this doesn't have any consequence on the performances on large-scale test cases, I would suggest going back to this value.

rosiereflo, 2022-01-26:
Yes agreed. We could maybe also add a warning if the precision of dynawo is higher than the solver one?

adriguir, 2022-01-28:
After further investigation, it seems most of the problems are due to inconsistent parameters and we can keep the code as it is (at least for addTerms).

closed 2022-01-28 by adriguir