dynawo/dynawo #2149: [Modelica Model compilation] Properly handle _event_floor modelica method

rosiereflo, 2022-02-02:


ErwanGuichard49, 2022-02-02:
With the improvement (2149_event_floor), I do have a segmentation error. The debug command leads to 

/home/guicharderw/Projects/dynawo/dynawo/dynawo/sources/Modeler/ModelManager/DYNModelManager.cpp:1416: double DYN::ModelManager::computeDelay(DYNDATA*, int, double, double, double, double): Assertion `delayManager_.isIdAcceptable(exprNumber)' failed.

Without the component FixedBooleanDelay, the simulation runs.

ErwanGuichard49, 2022-02-04:
With the creation of a Real variable uReal = if u then 1 else 0, the simulation does not run (segmentation error).

If I replace floor by integer in the calculation of yInteger (which is an integer), the compilation fails because of event_integer.

PR #2150 (2022-02-07): #2149 properly handle threadData and event_floor methods
closes #2149
commit bd32d7f8 (2022-02-07): #2149 properly handle threadData and event_floor methods
closes #2149