dynawo/dynawo #2156: BooleanDelay unit block

adriguir, 2022-02-04:
The "delay" keyword shouldn't be used for booleans because it works based on a linear interpolation with doesn't really make sense for a boolean (what about having a 0.5 value somewhere). 

Another satisfying way to implement this should be created.

adriguir, 2022-02-04:
In the ModelicaStandardLibrary version 4, there is a new block in the Logical package called LogicalDelay that seems interesting to do what we want to do.
It is not directly usable at the moment as the MSL version 4 is not yet supported neither in OM nor in Dynawo but can serve as a source of inspiration on a possible implementation to do in Dynawo. 

Of course, this has to be formally assessed and after double thinking.

cossartque, 2022-02-04:
> LogicalDelay

If I understand it right, this block LogicalDelay is very similar to OnDelay (and OffDelay) that are already in the MSL (and in Dynawo). The same problem as when using "when" arises: when we have multiple changes within tDelay then we lose some information.

adriguir, 2022-02-04:
You are completely right. I haven't thought about the multiple changes case. 
Let's see if we find another solution in Modelica, instead of using with real/boolean conversion and so on.

ErwanGuichard49, 2022-02-04:
I agree. We could declare a vector tSwitch, modify a component at each change of input. Drawback : the number of changes during tDelay would be capped.

cossartque, 2022-02-04:
> I agree. We could declare a vector tSwitch, modify a component at each change of input. Drawback : the number of changes during tDelay would be capped.

This is equivalent to having a buffer. It is doable but requires having a fixed maximum size of buffer vector. Moreover I think the performance would be poor.

ErwanGuichard49, 2022-02-07:
I have pushed a version with the delay function applied to a non-filtered uReal.

I have also replaced the floor function with a threshold crossing in order to trigger an event when the boolean output changes. With the new modifications about event_floor and event_integer, this substitution may be unnecessary.

ErwanGuichard49, 2022-02-07:
With the aforementioned threshold set to 0.5 i.e. halfway between 0 and 1, the blocks works according to expectations.

closed 2022-11-23 by cossartque