dynawo/dynawo #2170: Min/max handling

adriguir, 2022-02-10:
Min and max operators are continuous but not differentiable.
In the current way OpenModelica works, an if condition is generated for a min or a max condition without any associated root or mode. This leads to potential completely wrong differentiable value in the solving process certainly leading to failures at the numerical side.

In traditional OM simulations, there is no problem because in most of ode simulations, the min/max ends into assignment parts for algebraic variables and not into a system resolution.

adriguir, 2022-04-08:
The best option is certainly to keep at the moment the actual behavior. In latest versions of OM Compiler, a C++ min / max operator is generated, which is not perfect in terms of numerical behavior but is very helpful in case of sqrt structure later on.
Let's see later on what we do with this

rosiereflo, 2022-05-13:
The fact that an if condition is generated for a min/max is in fact the result of the patch applied to OpenModelica in Dynawo, so we could make some tests by removing this patch

adriguir, 2022-05-13:
We can indeed. Then we will end up with a call to the C++ min/max function I suppose. 
This will help us in the case of sqrt or any operator not defined on the complete R space because we will be sure never evaluating a sqrt of a negative quantity. That's the good point.
The negative point is that we will switch from one form of an equation to another without any warning or specific scheme (it will be a silent change).

=> Anyway, in the current state, we need to live with the negative point as we don't have better solution. Thus that will be nice to get rid of the patch and end up with the min/max Cpp function into the Cpp code.

closed 2024-02-16 by rosiereflo