dynawo/dynawo #2172: Delete the intermediate angle variable associated into the injector Idq model

adriguir, 2022-02-11:
This causes problems in dae mode when solving the problem because the associated angle is instantaneously switching from Pi to -Pi.

cossartque, 2023-05-12:
In order to delete the variable UPhase, calculated as the arg of uPu, I propose that UPhase becomes an input of the injector model. Then of course, whenever InjectorIDQ model is used, a proper PLL must be added to the model to provide this UPhase to the injector. For the moment InjectorIDQ is used in the WECC models and in the HVDC VSC models.

adriguir, 2023-07-14:
Test cases for the Wind Turbine alone (without the plant controller) have to be updated - they perhaps were not there at the time the development for this issue began. 
Parameters for the PLL should be filled, otherwise the behavior is quite erratic.

closed 2023-07-17 by joyelfeghali