dynawo/dynawo #2208: Initializing MSL TransferFunction

ErwanGuichard49, 2022-03-04:
Two issues with the initialization of TransferFunction block from Modelica Standard Library :

1) The x_start parameter is not taken into account by Dynawo, one needs to add a modifier for a proper initialization :+1: 
**x_scaled(start = {Y0})** instead of **x_start = {Y0}**
This may well be caused by the nature of x : it is an output whereas x_scaled is a (protected) variable.

This shortcoming is thus an inconvenience to the user.

2) The _x_scaled(start_ modifier has no impact if the start value is a vector of two components or more. In fact, with **x_scaled(start = {0, Y0})** combined with **x_start = {0, Y0}**,  the block x variable starts at 0.

This issue thus prevents Dynawo from initializing correctly a MSL block.

rosiereflo, 2022-03-04:
Any testcase available?

ErwanGuichard49, 2022-03-04:
nrt/data/SMIB/SMIB_GoverEdfTurHcVRSepGt is a relevant test case. TransferFunction is used in ExTkjSimp.mo, which is involved in the test case. The exact line is :

Modelica.Blocks.Continuous.TransferFunction leadLag(a = {tLag, 1}, b = {tLead, 1}, x_scaled(start = {Efd0Pu}), x_start = {Efd0Pu}, y_start = Efd0Pu);

A failure to initialize is visible when y and u are not identical at t = 0 s.

ErwanGuichard49, 2022-03-07:
Under OpenModelica in standard use, the issue 1 does not arise since there are several Init types available. Under Dynawo, the initial equations are ignored, thus compelling us to workarounds like x_scaled(start = {Y0}).

The second issue remains to be solved.

PR #2216 (2022-03-10): #2208 fix initialization of tables
closes #2208
commit deb5ef69 (2022-03-10): #2208 fix initialization of tables
closes #2208
closed 2022-03-10 by adriguir