dynawo/dynawo #2217: [NRT] Add the possibility to plot the reference curves

rosiereflo, 2022-03-09:
Add the possibility to plot the values from the reference curve files

rosiereflo, 2022-03-09:
@ErwanGuichard49 FYI

rosiereflo, 2022-03-17:
This feature already exists: myEnvDynawo.sh curves-reference <PATH TO JOBS FILE>

ErwanGuichard49, 2022-03-17:
On my computer, with this line, I get
myEnvDynawo.sh : commande introuvable

Could you show me your myEnvDynawo.sh file ?

Moreover, does this feature exist for dynawo-rte ?

ErwanGuichard49, 2022-03-17:
This feature exists for dynawo-rte and it works for me, but I would like to plot both the reference and the current output for comparison purposes on the same plot.

rosiereflo, 2022-03-17:
> This feature exists for dynawo-rte and it works for me, but I would like to plot both the reference and the current output for comparison purposes on the same plot.

got it

PR #2258 (2022-04-12): #2217 Add the possibility to plot the reference curves on the same graph
as the curves
closes #2217
commit 266b0fb5 (2022-04-12): #2217 Add the possibility to plot the reference curves on the same graph
as the curves
closes #2217
closed 2022-04-12 by gautierbureau