dynawo/dynawo #2246: [Criteria] All failing criteria should be added into the failing criteria list

rosiereflo, 2022-03-31:
Currently within a single criterion targeting several  components (bus, loads or generators) we stop the checking loop at the first failing component

adriguir, 2022-04-08:
A small additional remark too: in the case of a criterion like (U < 0.8 Un for all buses), it would be nice to output the bus with the lowest voltage and not the first one.
I don't remember if it is already the case or not. If yes, just forget about the remark!

rosiereflo, 2022-04-08:
No it's not the case, today we give the first failing bus by alphabetical order.
Just to be sure that I got your point: you mean ordering the buses in increasing order?

rosiereflo, 2022-04-19:
I propose that we print:
- voltage constraint: the 10 (or whatever) worst nodes in the timeline and all nodes in the log (DEBUG filter);
- power constraints: the final constraint in the timeline and the full list of participating sources in the log (DEBUG filter).
would that be ok?

adriguir, 2022-04-19:
Quite fine for me. I would just suggest to output a bit less worst nodes in the timeline (5 worst?)

closed 2022-11-15 by rosiereflo