dynawo/dynawo #2264: BooleanArray and .extvar

ErwanGuichard49, 2022-04-12:
I would like to add a Boolean vector input to an .extvar file.

The only use of an array in a .extvar in the whole library is in _dynawo/sources/Models/Modelica/Dynawo/Electrical/Controls/Generic/GenericAutomaton.extvar_

<variable type="continuousArray" id="inputs" size="100" optional="true"/>

I have tried to write a similar line with a **discreteArray**, the compilation fails.

ErwanGuichard49, 2022-04-22:
The branch 472 of GitLab would compile if this issue were resolved.

rosiereflo, 2022-04-29:
The discreteArray support seems to be already implemented.
However we need to have coherency between the table size in the .mo and the extvar. I was able to compile by assigning '4' to the size attributes of the discreteArrays in the branch + using the branch containing the fix for the issue #2263