dynawo/dynawo #2408: BooleanPulse : segmentation error

ErwanGuichard49, 2022-06-29:
On the branch **2363_combitable**, a model BooleanPulse.mo has been added to _Dynawo.Electrical.Controls.Basics_, it is based on the MSL block of the same name. The compilation is successful.

On the branch **501_add_svarc_ab2**, the test case dynawo-rte/nrt/data/SVarC/SVarC_ABB/SVarC.jobs uses this component. The compilation of StaticVarCompensatorABB is successful but the simulation is not started. I get the following error :

_/home/guicharderw/Projects/dynawo/dynawo/deploy/gcc9/shared/dynawo/bin/launcher : ligne 174 : 302332 Erreur de segmentation  (core dumped) ${DYNAWO} ${ARGS}
Dynawo job failed._

If I replace BooleanPulse by BooleanTable, the simulation runs unencumbered.

The MSL block BooleanPulse involves a discrete variable _pulseStart_ and the _sample_ function.

rosiereflo, 2022-07-05:
Renaming this ticket as the sample feature is not supported in Dynawo

cossartque, 2022-09-22:
What's the current status ? Is it solved ?

ErwanGuichard49, 2022-09-22:
It remains unsolved.